spth/OpenRabbit #1: Cold-boot only mode?

spth, 2020-08-30:
Currently, OpenRabbit does
1) Use cold-boot mode (slow - 2400 bps, load to RAM, lower 64 KB only) to load coldboot.bin (a binary blob from dynamic C that implements a medium-speed (57.6 bps or 19.2 bps) loader)
2) Use the medium-speed loader to load pilot.bin (a binary blob that implements a fast loader)

The main disadvantage of this process is the requirement for the two binary blobs from non-free Dynamic C.

Exposing a cold-boot only functionality could allow to write a program onto the Rabbit without needing any binary blob.

tomlogic, 2020-09-11:
Cold-boot only will be agonizingly slow and can only address the lower 64K of memory.  Every byte of the cold loader actually requires three bytes (16-bit address and 8-bit data byte) over the serial port, so at 2400bps you'll only get 80bytes/second of throughput.  In addition, the written program will just be in RAM.

You could write a new pilot program with necessary code to receive the full firmware image and write it out to flash (or RAM).  I believe the cold loader and pilot source describe the method used.  You could also check Dynamic C 10 documentation as it may be more thorough (and describe a bootstrapping method that would apply to writing a boot strap process for Rabbit 2000/3000 products from scratch).

It might be possible to copy C code from the Open Source Dynamic C libraries and BIOS to create a .C program that you can compile from SDCC to create a fully-free pilot.c.

tomlogic, 2020-09-14:
But, after looking at what the Pilot BIOS can do, I could see implementing a "Pilot-only" mode.  You could read or write any location in memory in that mode, including reading the System ID Block.  I'm curious about using it to download a copy of the installed firmware, something that should be possible on any device with just the Pilot BIOS.

spth, 2020-09-21:
Indeed cold-boot only would be very slow. And since we have free coldload.bin and pilot.bin there not that much motivation anyway (though we still can't compile pilot.bin using free tools, so maybe an initial loader only mode would make sense).

closed 2020-09-21 by spth