spth/OpenRabbit #2: Free replacements for coldboot.bin and pilot.bin

spth, 2020-08-30:
OpenRabbit currently depends on the coldboot.bin and pilot.bin binary blobs from Dynamic C. It would be good to have free replacements.

spth, 2020-08-30:
This would essentially be done by translating the Dynamic C source at https://github.com/digidotcom/DCRabbit_10/tree/master/ColdBoot to sdas syntax (despite being named .C these are actually assembler source files). Then we could compare the resulting binary to the coldboot.bin and pilot.bin that come with Dynamic C.

spth, 2020-08-30:
At least the source for pilot.bin from Dynamic C 10 won't work, as it uses Rabbit 4000-specific instructions that won't work on Rabbit 2000 and 3000. Need to check coldboot.bin (and see if pilot.bin source from Dynamic C 9 will be released as free software).

spth, 2020-09-01:
Source for pilot.bin and coldboot.bin suitable for Rabbit 2000 and 3000 from Dynamic C 9 has just been released under MPL:
https://github.com/digidotcom/DCRabbit_9/commit/d939c1440e76c2e505c46679b73bef45be483db2

spth, 2020-09-08:
The coldoad branch now has coldboot.bin source that can be compiled with SDCC.
Doing the same for pilot.bin would be possible, but more work.

tomlogic, 2020-09-11:
It should be safe to remove the `nop` opcodes and trailing instructions.  As mentioned in our other communications, I believe the `nop`s are placeholders for debugging via `rst28` opcodes and unnecessary, and the 5 instructions at the end are unused/unnecessary.

spth, 2020-09-12:
I intent to remove them later, once I have the RFU-equivalent functionality working with OpenRabbit.

spth, 2020-09-14:
In current trunk, the unneeded instruction in coldload.bin are removed. It still works fine.

spth, 2020-09-15:
Replacing pilot.bin will be harder. The source is mostly assembler code with C declarations. SDCC is a great C compiler, but it doesn not support as much interaction between C and assembler as Dynamic C does. This will be more of a rewrite than a simple syntax translation.

feilipu, 2023-12-24:
Philipp, I've taken on a [RCM2024](https://github.com/feilipu/rcm2024) project to build a cheap (hobby) version of a Rabbit 2000 board, since there's nothing else around, afaik.

As a starting point I've been donated a RCM4300 which I can use to play with, and get the loaders and crt0 working with z88dk and sdcc.

Since Dom and Paulo have added R4000 mnemonics to z88dk, I can rewrite the Dynamic C cold boot and pilot code in native assembly.

So just wondering whether you're still interested in Rabbit stuff to add some advice to my endeavour?

Merry Christmas.

spth, 2023-12-28:
Well, I still maintain the Rabbit ports (r2k, r2ka, r3ka) of SDCC. Having a reliable and fully free software way to test on hardware would be good. I do have some RCM, so hardware is less of an issue for now.
We do now have a free (with source) coldboot.bin. We currently do not have source for a pilot.bin. However, IMO the current main issue are some sporadic hangs I see when using OpenRabbit.
I'm not near my Rabbit hardware for the moment, so I can't do any testing this or next week.

feilipu, 2023-12-28:
Thanks Philipp.