spth/OpenRabbit #4: Hangs on reading pilot checksum

spth, 2020-09-11:
I got a Digi 2 mm USB programing cable (says "This programming cable requires Dynamic C v 9.62 or later" on it).
Since I don't have the prototyping board, I just connect the RCM2200 to a lab power supply to supply and the programming cable.

I use the coldboot.bin and pilot.bin from Dynamic C 9.62.
I invoke Openrabbit via:

./rfu /home/philipp/OpenRabbit/coldboot/coldload.bin
/home/philipp/OpenRabbit/coldboot/pilot.bin test test test test /dev/ttyUSB2

But both in my (here) and the unchanged code from the latest release (01.1.), it hangs when trying to read the pilot checksum. I verified this by enabling #ifdef DEBUG_COMM in myio.c:

~~~~
sending 46477 pilot
writing 46477 bytes
reading 2 bytes
~~~~

OpenRabbit never proceeds beyond that last line.

At this point, there are multiple potential underlying issues: A bug in OpenRabbit, and incompatibility with the cable, an incompability with the pilot.bin.

tomlogic, 2020-09-11:
Can you capture serial communications and compare it to serial communications of RFU sending the cold loader and pilot?

spth, 2020-09-12:
I just checked that the hardware is okay: The same setup (same cable, same module, no prototyping board, lab power supply at 5V) works fine with the RFU from Dynamic C 9.62A on Windows 10.

spth, 2020-09-12:
I'll look into capturing the communication later: I'll be away from most of my equipment including the lab power supplies and the Windows machine for a week from today.

spth, 2020-09-12:
I wonder if this is due to some difference between the coldbload.bin and pilot.bin that OpenRabbit was used with at the time of the 0.1.1 release vs. the current ones.

AFAIK the author of Openrabbit 0.1.1 used it with pilot.bin and coldload.bin from Dynamic C 8.51. Are pilot.bin and coldload.bin from Dynamic C 8.51 available somewhere? Do we know anything about the differences to the current versions?

spth, 2020-09-13:
Testing with coldload.bin and pilot.bin from Dynamic C 8.61 I was able to proceed beyond this problem.
So this is indeed a version incompability.

tomlogic, 2020-09-13:
I have documented the bootstrapping process from Dynamic C 9.62 in commit https://github.com/digidotcom/DCRabbit_9/commit/4e527a939a54a77e0a0eca342690978b13d1adad.

spth, 2020-09-14:
Turn out I was able to get beyond this exactly once so far. Now I run into https://github.com/spth/OpenRabbit/issues/5 even when powering off the RCM2200 overnight.

spth, 2020-09-14:
With the timing fixes, it works reliably using the Dynamic C 8.61 binaries.
It also works with our coldload.bin (from Dynamic C 9.62), but not with the pilot.bin from Dynamic C 9.62.

tomlogic, 2020-09-14:
My guess is that `PILOT.BIN` from DC 8.61 used a different protocol than the `PILOT.BIN` from DC 9.62.

spth, 2020-09-14:
Looks like it. Just using the pilot.bin from Dynamic C 9.62A fails due to different file format of pilot.bin.
But even when I change OpenRabbit to use the DC 9 pilot.bin file format, it still fails:

~~~~
reset rabbit
loaded 831 bytes from ../coldboot/coldload.bin
flushing data for baudrate set
set baudrate to 2400
sending 828 initial loader triplets
flushing data for baudrate set
set baudrate to 57600
loaded 5517 bytes from /tmp/pilot.bin
sending 5517 pilot
csumR 0xd469 != csumU 0x0000
~~~~

spth, 2020-09-14:
It works now. With pilot.bin from Dynamic C 9 (which is included in OpenRabbit).
After all it was just a file format issue.

closed 2020-09-14 by spth

tomlogic, 2020-09-14:
OK, so now it sounds like the coldboot/pilot from Dynamic C 8.61 and 9.62 are interchangeable with the protocol used in OpenRabbit's RFU replacement?  That's great.  And you have a way to assemble `coldboot.s` with SDCC, and convert it to triplets for a replacement `coldboot.bin`.  Next would be to update `pilot.c` and its dependencies to compile with SDCC (perhaps a bit more difficult, given Dynamic C's library format).