spth/OpenRabbit #5: pilot.csum 0x4a != csum 0x00 on second try

spth, 2020-09-13:
When trying to use OpenRabbit a second time I usually see the following problem:

~~~~
pilot.csum 0x4a != csum 0x00
~~~~

pilot.csum of course depends on pilot.bin (the above is for the one from Dynamic C 8.61), but the error is always the same.

My setup is an RCM2200 supplied with 5 V from a lab power supply and connected to a USB programming cable from Diig. This hardware setup works on Windows 10 with the RFU from Dynamic C 9.62A, on the first and further tries. And using the coldload.bin and pilot.bin from Dynamic C 8.61 it works on the first try on Debian GNU/Linux. But not on subsequent ones.

So far the only workaround is to disconnect the RCM2200 from the power supply and programming cable and wait a few minutes. Waiting just one or two minute isn't enough.

spth, 2020-09-14:
Current trunk now check the status line before and after sending the initial loader.
Turn out that something goes wrong when sending the initial loader: The status line is fine before sending it, but it doesn't go high after sending the loader (but if I omit the actual sending of the loader, the byte sequence sent after, that is supposed to set the line high, works fine).

tomlogic, 2020-09-14:
Hmm.  2400bps is 240 bytes/second or 80 triplets/second, which is 12.5ms.  So your delay aligns well with that.

I haven't looked at the code, but what happens if you place your entire delay after sending the complete loader?  Just dump the triplets and then wait 3.5 seconds (or whatever the byte count * 5ms comes to) before timing out on your status line check?

My guess is that the send happens instantaneously and the OS buffers the bytes.  You may need to check the serial port's transmit buffer and wait for it to drain before expecting the status line to change.

spth, 2020-09-14:
True. It also works if I just wait the whole time period in the end.
I don't know of a way to flush the buffer (and with an USB converter involved, there might not be even a way to flush the whole buffer); so for now I'll try to make stuff work with the delays.

spth, 2020-09-14:
And once I fixed all the details, it now works reliably using the loader binaries from Dynamic C 8.61!
Waiting isn't elegant, but we can keep it that way until we find a better solution.

closed 2020-09-14 by spth

tomlogic, 2020-09-14:
You could update the code to watch for the STATUS pin to change, and base the timeout on the number of bytes sent.

This [Stack Overflow answer](https://stackoverflow.com/a/30850731/266392) has a possible method to determining how many bytes are still in the output buffer.  But as you said it might not be accurate due to buffering in the FTDI chip.