spth/OpenRabbit #7: Make it cross-platoform

spth, 2020-09-14:
OpenRabbit currently uses Linux-isms. Since it requires direct access to DTR, etc it can't just use the portable POSIX functionality instead. Also, it would be nice to have Windows support.
Thus a serial port abstraction layer is needed. Options include:
* Roll our own, like https://github.com/free-pdk/easy-pdk-programmer-software does
* Use an existing library like the one from the sigrok project https://sigrok.org/wiki/Libserialport

tomlogic, 2020-09-14:
I wrote a cross-platform (Windows/POSIX) serial port library for https://github.com/digidotcom/xbee_ansic_library.  It includes control over RTS and reading of CTS, but I did not implement an API to control DTR.  I imagine you could extend it with an ioctl on POSIX and the appropriate `EscapeCommFunction()` on Windows (like I did in `xbee_ser_set_rts()`).

But libserialport looks like a more complete solution.

spth, 2020-09-15:
As far as I know, POSIX does not support explicit control of DTR. So for DTR, we'd need  different solutions for individual POSIX systems (like the Linux ioctl Openrabbit uses now), though I think FreeBSD and OpenBSD are very similar to Linux wrt. the API for DTR.
An advantage of libserialport is that it is already is part of Debian, other GNU/Linux distributions, and in the OpenBSD and FreeBSD ports.
However, AFAIK, libserialport does not yet support Hurd and I don't know of libserialport Windows packages. Also, while libserialport is under active development (and maintained by the sigrok project, so it won't go away anytime soon), they haven't made a release in years.