spth/OpenRabbit #8: Intel hex support

spth, 2020-09-15:
OpenRabbit should support the use of Intel hex (.ihx, .hex) files. It should try to detect if a supplied file is Intel hex vs. binary (maybe just checking the filename like other software does would be enough).

Intel hex is a common format. It is also the default output format of the Small Device C Compiler (SDCC), the main alternative to Dynamic C for Rabbit development.

tomlogic, 2020-09-24:
One benefit would be the ability to skip over unused memory ranges.  It might be possible to optimize the `.bin` loader to ignore blocks of all `0xFF`, since that's what the flash erases to (I believe that's true for all of the products).

Best to integrate an existing Open Source `.ihx` parser instead of implementing from scratch.

spth, 2020-09-25:
Intel hex parsers are small, and I don't know of a commonly used library.

I was considering just copying the one from stm8flash (https://github.com/vdudouyt/stm8flash) into OpenRabbit.

After all we know that that one handles Intel hex files from SDCC well.

spth, 2021-04-16:
For project files, this is now implemented, using the ihex.h and ihex.c from stm8flash.