spth/OpenRabbit #15: Auto-detect secondary loader format

spth, 2020-09-20:
OpenRabbit currently has the --dc8pilot option for the use of a DC8-style secondary loader.

It could make sense to instead auto-detect the secondary loader style, so the user doesn't have to specify this explicitly.

tomlogic, 2020-09-20:
I like the command-line option, and I don't think there will be a reliable way to identify which Pilot BIOS protocol is in use by a given `pilot.bin` file.

spth, 2020-09-20:
Well, the secondary loader is placed at 0x6000.
A DC9 pilot.bin contains data to be placed from there. It starts with the first instruction, which typically is nonzero.
A DC8 pilot.bin contains the data as seen in memory later: Data at addresses 0x0000 to 0x5fff in a DC 8 pilot.bin is all zero.

tomlogic, 2020-09-20:
Good point.  I noticed that discrepancy in some of the older DC 9 `pilot.bin` files (or maybe the "ram-only" version).  I am assuming that RFU and Dynamic C had code to skip over those opening 0x00 bytes, but at some point the build process for `pilot.bin` was able to exclude the extra 0x00 bytes at the start of the file.

spth, 2020-09-23:
Implemented in 527d184.

closed 2020-09-23 by spth