spth/OpenRabbit #20: Display human-readable processor information

spth, 2020-09-23:
OpenRabbit should display the type of processor detected in human-reaaable format ("Rabbit 2000C", etc) instead of just a hexadecimal number.

spth, 2020-09-23:
Implemented in 706ec71.

spth, 2020-09-23:
For now I implemented this only for the various Rabbit 2000 and 3000 variants. Other Rabbits will show as "Unknown processor". According to the user manuals, the Rabbit 4000, 5000 and 6000 all have the same value in GCPU and GREV registers. So the proccessor type is 0x0200 for all of them.

tomlogic, 2020-09-23:
I'm guessing that the 4000-6000 will require the pilot bios from DC 10, which will have support for the different memory configurations.

tomlogic, 2020-09-24:
We should probably close this out.  Adding IDs for later processors would be a part of adding support for those processors.  If you check the DC 10 documentation in the Coldboot directory, there's a more detailed device identification process (beyond just coldload/pilot).  Some devices have 16-bit memory and some have serial boot flash instead of parallel flash.

spth, 2020-10-05:
In de63287 all processors are reported (assuming manual section 4.4 is right and section 4.1.2 is wrong - see https://www.digi.com/support/forum/76098/how-to-tell-the-difference-between-rabbit-4000-5000-and-6000?show=76248#c76248).
Even if we don't support or haven't tested with Rabbit 4000, 5000, 6000, if someone reports an issue, it might still be helpful, if we get a specific processor name rather than "Unknown processor".

closed 2020-10-05 by spth