spth/OpenRabbit #21: Low Dhrystone performance

spth, 2020-09-23:
On the RCM2200, which has a Rabbit 2000@24 MHz, I see 2755 Dhrystones/s using the example code with current SDCC.

For comparison, current SDCC gets about 6000 Dhrystones/s on a C8051@96 MHz, and over 10000 Dhrystones/s on a STM8AF@24 MHz.

spth, 2020-09-23:
I've checked the memory setup. Looks like there are no wait states, so nothing to be gained there.
The clock doubler is apparently active, too, so we're running at full speed.
I've read the generated assembler code, and was unable to identify a single big problem. Still, I've found about a dozen places with potential for improvement, and will look into improving SDCC accordingly. This should improve the score a bit, but I don't think we'll be able to reach 3000 Dhrystones/s on a 22 MHz Rabbit 2000.

tomlogic, 2020-09-25:
Dhrystone on RCM3209 (44Mhz) had a bunch of failures and printed some garbage, but resulted in:
```
Microseconds for one run through Dhrystone: 238                                 
Dhrystones per Second:                      4190                                
```

Note that I had to add `-DNOSTRUCTASSIGN` to `SDCCFLAGS` to get it to compile.  Depending on how much memory it makes use of, there may be conflicts with the RAM region starting at 0x8000 overlapping with the stack at 0xD000. 

Output from running Whetstone on RCM3209 (44Mhz):
```
Loops: 10, Iterations: 1, Duration: 33006 msec.
C Converted Double Precision Whetstones: <NO FLOAT> KIPS
```

spth, 2020-09-25:
Your Dhrystone score make sense to: compared to mine, your setup has twice the clock rate, but needs additional wait states.
Also, you use an old SDCC version (as we can see from the need for -DNOSTRUCTASSIGN) that probably lacks some optimizations.
Similar for Whetstone, where I see

~~~~
Loops: 10, Iterations: 1, Duration: 49745 msec.
C Converted Double Precision Whetstones: <NO FLOAT> KIPS
~~~~

I.e. the same number of iterations, at less than twice the time.

spth, 2020-09-25:
We should real track down those failures.
For me, the values in Dhrystone look as they should:

~~~~
Dhrystone Benchmark, Version 2.1 (Language: C)

  Ptr_Comp:          -27638
        should be:   (implementation-dependent)
  Discr:             0
        should be:   0
  Enum_Comp:         2
        should be:   2
  Int_Comp:          17
        should be:   17
  Str_Comp:          DHRYSTONE PROGRAM, SOME STRING
        should be:   DHRYSTONE PROGRAM, SOME STRING
Next_Ptr_Glob->
  Ptr_Comp:          -27638
        should be:   (implementation-dependent), same as above
  Discr:             0
        should be:   0
  Enum_Comp:         1
        should be:   1
  Int_Comp:          18
        should be:   18
  Str_Comp:          DHRYSTONE PROGRAM, SOME STRING
        should be:   DHRYSTONE PROGRAM, SOME STRING
Int_1_Loc:           5
        should be:   5
Int_2_Loc:           13
        should be:   13
Int_3_Loc:           7
        should be:   7
Enum_Loc:            1
        should be:   1
Str_1_Loc:           DHRYSTONE PROGRAM, 1'ST STRING
        should be:   DHRYSTONE PROGRAM, 1'ST STRING
Str_2_Loc:           DHRYSTONE PROGRAM, 2'ND STRING
~~~~

Dhrystone uses about 5 KB of RAM. I wonder if we are running into one of the known Rabbit bugs (https://sourceforge.net/p/sdcc/wiki/Rabbits/), that only affect configurations with wait states. Which processor do you have (Rabbit 3000 vs. 3000A)?

spth, 2020-09-25:
To check if it is a wait state bug: reduce clock speed until you no longer need wait states, configure memory without wait states and see if the problem goes away.

Today, I implemented some workarounds for Rabbit wait state bugs in SDCC. If your problems turn out to be wait state bugs, please check if the problem is still there when using SDCC 4.0.3 #11878 or later

tomlogic, 2020-09-25:
Or it could be older bugs, I'm running the `sdcc` package from Ubuntu and have been considering replacing it with a self-compiled build:
```
SDCC : mcs51/z80/z180/r2k/r3ka/gbz80/tlcs90/ds390/TININative/ds400/hc08/s08/stm8 3.8.0 #10562 (Linux)
```

I assumed that's why I'm seeing this warning pop up frequently:
```
portme.c:22: info 218: z80instructionSize() failed to parse line node, assuming 999 bytes
'ioi'
```

Can you point me to which wait state bugs you implemented workarounds for?  I'm aware there were some issues with `LDIR` and similar opcodes.  For the record, I'm using a Rabbit 3000A.

spth, 2020-09-25:
I implemented workarounds for the mul, ldir and lddr ones (now lower-case letters at https://sourceforge.net/p/sdcc/wiki/Rabbits/). But according to the user manuals, none of them affects the Rabbit 3000A.

spth, 2020-09-27:
I managed to improve SDCC a bit, and Dhrystone score on the RCM2200 is now (SDCC 4.0.3 #11876) back up to 2752 Dhrystones / s.With some improvements in register allocation I hope to improve it a bit further, but for Dhrystone in total I don't think we can do much better.

The Rabbit 2000 and Dhrystone are just a bad match:

Dhrystone makes heavy use of 4 standard library string functions.
- memcpy - suffers from lack of useable ldir (wait state bug - fixed in Rabbit 3000A)
- strcpy with string literal parameter - suffers from lack of useable ldir (wait state bug - fixed in Rabbit 3000A)
- strcmp - there is a bit of potential for improvement in SDCC here, and I'll look into it later today
- strlen - suffers from lack of cpir (design choice - Rabbit Semi claimed they dropped it due to being "not very useful")
There also is an integer multiplication in Dhrystone, which suffers from lack of a useable mul (wait state bug - fixed in Rabbit 2000A)

For comparison, without the wait state bugs, current SDCC would reach 2907 Dhrystones / s on the same hardware (the RCM2200 doesn't actually use wait states and the r3ka backend happens to not emit any Rabbit 3000A - specific instructions for Dhrystone, so code compiled using -mr3ka can actually run on the RCM2200).

spth, 2020-09-27:
With today's changes in SDCC, the Rabbit 2000 on the RCM2200 now reaches 2809 Dhrystones / s. A Rabbit 3000A would reach 3070 Dhrystones / s at the same clock rate. A Rabbit 2000A (once the r2ka port has been implemented in SDCC) something in between.

In terms of Dhrystone performance, that looks good for this hardware enough to me, and I don't expect any further easy improvements.

tomlogic, 2020-09-28:
Can SDCC include an option to disable the workarounds for devices without wait states, or where the RAM and flash have the same number of wait states?  And would it be possible for an affected product's `crt0.s` or target header file include an `#error` statement if the workarounds had been disabled?

spth, 2020-09-28:
An `#error` in a header should be possible, but probably not in crt0.s.

A problem with the option is that the same standard library would be used with or without the option, so functions in the standard library would not benefit.

For now, I intend to create a r2ka port for SDCC in addition to the existing r2k and r3ka ports.
The manual would contain a note that when no memory needs wait states, code generated by the r2ka port can also run on the original Rabbit 2000.

The r2ka port would still be restricted in its use of ldir, but it could use mul.

spth, 2020-10-05:
Recent improvements in SDCC resulted in a further slight increase - for r2ka we are now at 2904 Dhrystones / s on that RCM2200@22MHz.
I don't think there is any low-hanging fruit left that would give big increases in Dhrystone.

closed 2020-10-05 by spth