spth/OpenRabbit #28: Doesn't work for Rabbit 5000/6000

spth, 2025-04-15:
I have tried OpenRabbit with the RCM5700 and RCM6700, but it hangs at

~~~~
philipp@notebook6:~/OpenRabbit$ src/openrabbitfu --verbose --verbose coldboot/coldload.bin coldboot/pilot.bin examples/hello/hello-RCM4110.bin /dev/ttyUSB3
Reset Rabbit.
loaded 831 bytes from coldboot/coldload.bin
Sending initial loader.
set baudrate to 2400
sending 276 initial loader triplets
set baudrate to 57600
loaded 5517 bytes from coldboot/pilot.bin
Secondary loader format detected as Dynamic C 9
Sending secondary loader.
^C
~~~~

By contrast, OpenRabbit works for me for various Rabbit 2000/3000/4000 boards.

spth, 2025-04-15:
I suspect that our initial loader's clock frequency detection does not work for the Rabbit 5000/6000, and thus the baud rate gets set incorrectly. We use a timing loop (just like Dynamic C 9 does):

~~~~
; MPL 2.0 code
wait_for_zero:
	ioi
	ld	(RTC0R), a                   ; fill RTC registers
	ioi
	ld	hl, (RTC0R)                  ; get lowest two RTC regs
	and	hl, de                       ; mask off bits
	jr	nz, wait_for_zero            ; wait until bits 0-9 are zero

timing_loop:
	inc	bc                           ; increment counter
	push	bc                           ; save counter
	ld	b, #0x98                     ; empirical loop value
		                             ; (timed for 2 wait states)
	ld	hl, #WDTCR
delay_loop:
	ioi
	ld	(hl), #0x5a                  ; hit watchdog
	djnz	delay_loop
	pop	bc                           ; restore counter
	ioi
	ld	(RTC0R), a                   ; fill RTC registers
	ioi
	ld	hl, (RTC0R)                  ; get lowest two RTC regs
	bit	2, h                         ; test bit 10
	jr	z, timing_loop               ; repeat until bit set
~~~~

In the delay_loop, we have `djnz`, which takes 5 cycles on Rabbit 2000/3000/4000, but takes 6 cycles on Rabbit 5000/6000.
We could try modifying the loop to use `dec b`, `jp nz, delay_loop` since that sequence has the same timing on all Rabbits (as long as 8-bit memories are used), or try to switch to a timer-based approach, like Dynamic C 10 does.

spth, 2025-04-15:
Hmm. I've tried a jp-based loop, but the result is still the same - works for me on Rabbit 4000, but not Rabbit 5000:

~~~~
	ld	hl, #WDTCR
	ld	b, #0x73                     ; empirical loop value (timed for 2 wait states)
delay_loop:
	ioi
	ld	(hl), #0x5a                  ; hit watchdog
	dec	b
	jp	nz, delay_loop
~~~~

tomlogic, 2025-04-15:
I think the documentation about [Dynamic C 10's "cold boot"](https://github.com/digidotcom/DCRabbit_10/tree/master/ColdBoot) will help.  You should make use of the cold loader and pilot BIOS files that were written for the 4000/5000/6000.

spth, 2025-04-15:
I'm aiming for automatic support of Rabbit 2000 to Rabbit 6000, so the user doesn't have to specify which Rabbit is connected. I see that Dynamic C 10 does detection of Rabbit 4000 vs. Rabbit 5000/6000 by setting the lowest two bits of MB0CR to 3 (to check for the internal RAM of Rabbit 5000/6000). I guess being able to detect internal RAM would be a good next step forward OpenRabbit, too. Then we setup RAM, then send our old coldload.bin (which will only work for 8-bit memory and at most 1 MB of memory, and parallel memory only, but that should already get us to RCM5700 support). To support RCM with serial flash or 16-bit memories, or more than 1 MB of total memory, we'd then need to replicate more of the Dynamic C 10 bootstrapping.

According to the documentation, setting the lowest two bits of MB0CR to 3 has the following effect:

Rabbit 2000 - use /CS2
Rabbit 3000 - "reserved and should not be used"
Rabbit 4000 - "reserved and should not be used"
Rabbit 5000 - use /CS3 (i.e. internal RAM)
Rabbit 6000 - use /CS3 (i.e. internal RAM)

Since Dynamic C does it on the Rabbit 4000, I guess we can assume it to be safe there. That leaves the Rabbit 3000 as a possible problem, but I'll see what I'll find.

Colonel-Bleep, 2025-04-15:
I don’t think that dc10 supports 3000 and below.  Needs dc9, unless something changed.

tomlogic, 2025-04-15:
At one point, Dynamic C 10 was going to support the 2000/3000 products.  You can see that legacy in the "Lib" directory having a separate `Rabbit2000_3000` subdirectory that wasn't removed until the 10.72 release.

Note that the RFU program from Dynamic C 10 supports all of the Rabbit processors, but you need to tell it whether you're doing 2000/3000 (and using the DC9-based loader/pilot) or 4000/5000/600 (and using DC10-based loader/pilot).

spth, 2025-04-16:
> I don’t think that dc10 supports 3000 and below.  Needs dc9, unless something changed.

My personal vision for openrabbitfu is as a simple, free tool to write and read the flash memory of all Rabbits, that works well on modern OSes. In the long term, I'd also see a tool for on-target debugging with gdb on all Rabbits (which might be openrabbit or something else, e.g. OpenOCD-based). I see Openrabbit as part of a free toolchain targeting the Rabbits. Since SDCC will make substantial steps towards better Rabbit support this year, OpenRabbit should try to keep up a bit, and adding Rabbit 5000/6000 support is part of that.

IMO, a good first step will be to allow the user to select the RAM location via a command-line option, then autodetection of the RAM location (CS0 vs. CS1 vs. CS2 vs CS3). This should already allow us to write the Flash on the RCM5700, RCM5710, RCM5750, RCM5760 and read out the processor type and revision on all Rabbits (unless they use 16-bit RAM).

At that point, we'd still use the pilot from Dynamic C 8 / 9, and thus would be limited to 1 MB address space, 8-bit parallel memories. Our coldload.bin might have evolved away from the Dynamic C 8 / 9 one, and might look more like the Dynamic C 10 one, or something different. Next, we'd want our coldload.bin to either support 16-bit memories or have a separate one for 16-bit memories (like Dynamic 10 does), so we can also read out the processor type and revision on Rabbits with 16-bit RAM (e.g. RCM4000 and RCM4010). This however is a less important feature, since there are currently no plans for SDCC to support 16-bit memories.

Once we've reached that point, the obvious next steps would be support for an address space larger than 1MB, and support for serial flash. But serial Flash would also need support on the SDCC side.

To get beyond the 1MB boundary, for 16-bit memory support, for serial flash support, we might use the pilot from Dynamic C 10 or write our own replacement.

spth, 2025-04-21:
A little bit of progress: I have the cold / initial loader, and to some degree the pilot / secondary loader working on the RCM5700:

~~~~
philipp@notebook6:~/OpenRabbit$ src/openrabbitfu --run --serialout --ramcr 0x43 examples/hello/hello-RCM5700.bin /dev/ttyUSB0
Sending initial loader.
Secondary loader format detected as Dynamic C 9
Sending secondary loader.
CPU:  0x0300 (Rabbit 5000)
Freq: 25000000
sectorSize: 0x0080
numSectors: 0x0013
flashSize:  0x0080
~~~~

It then fails to upload the hello-RCM5700.bin, though. And on an RCM6710, it doesn't even get that far:

~~~~
philipp@notebook6:~/OpenRabbit$ src/openrabbitfu --verbose --run --serialout --ramcr 0x43 examples/hello/hello-RCM5700.bin /dev/ttyUSB0
Reset Rabbit.
Sending initial loader.
Sending 276 initial loader triplets.
Secondary loader format detected as Dynamic C 9
Sending secondary loader.
pilot.csum 0xe2 != csum 0x90
~~~~

Still, this is some progress, and once I've cleaned up the code a bit, and tested that the changes don't break Rabbit 2000 - 4000 support, I'll commit it.

tomlogic, 2025-06-03:
Here's the schematics for the RCM5700, in case you find that helpful.

[RCM5700 Schematic (090-0268).pdf](https://github.com/user-attachments/files/20573083/RCM5700.Schematic.090-0268.pdf)

You could compare the bootstrapping sequence from Dynamic C 10 to the one from Dynamic C 9 to see if there are key differences.

Doesn't the Rabbit 5000 chip include built-in RAM?  Check the data sheet on that.  I think bootstrapping makes use of it for at least part of the process.  It might not be large enough to hold the full Pilot BIOS.

spth, 2025-06-03:
Thanks. I'll check the schematic, to see if it offers any hint at what might go wrong. The Rabbit 5000 has built-in RAM, and the RCM5700 has no other RAM. The amount of internal RAM is 128K, i.e. the same as the external RAM on the RCM2020 board, which works with OpenRabbit.

spth, 2025-06-14:
While I still have hope for the Rabbit 5000, I don't think the current approach would work for the Rabbit 6000: the Rabbit 6000 internal RAM is apparently 16-bit mode only, and I think using 16-bit mode would require substantial changes.

spth, 2025-06-19:
I'd also like to make writes to the RCM5700 8-bit parallel flash work before looking into 16-bit memories. I've tried to read out the first 512KB of the Flash I'm trying to write with OpenRabbit, and it looks like it is all 0xff (except for a few dozen bytes at the end), both before and after the write attempt.
Also, the flash info reported looks a bit odd to me (same data as before):

~~~~
CPU:  0x0300 (Rabbit 5000)
Freq: 25000000
sectorSize: 0x0080
numSectors: 0x0013
flashSize:  0x0080
Flash speed: 70 ns
RAM speed: 15 ns
~~~~

19 sectors matches the data sheet. But the size doesn't make sense (no single size would make sense - according to the data sheet the Spansion AL008D chip on the RCM5700 has sectors of 4 different sizes. Also, the total flash size according to RCM5700 documentation and Spansion datasheet is 1MB, not 128 KB.
According to the datasheet, this flash should erase to all 0x00 (instead of the AFAIK common all 0xff), so I'm not sure if the write itself fails, or it is actually the preceding erase that isn't working.

spth, 2025-06-20:
I encountered something weird on the RCM5700. When I try to send a certain byte sequence to the coldloader, something goes wrong - either I get a wrong checksum back, or none at all, i.e. the cold loader apparently hangs. And once this is triggered, even after a reset, and reupload of the coldloader certain individual bytes will rettrigger the hang / wrong checksum if sent as first byte of the secondary loader (none of these bytes trigger the issue by themselves after a power-on-reset followed by upload of the coldloader).
The sequence is apparently 1380 bytes long (might be possible to remove a few bytes from the beginning or end, but less than 10 each), and I have not yet found a second one that triggers the issue, but this sequence is part of the Dynamic C 10 pilot. If I make minor changes to the coldloader (e.g insert or remove a nop somewhere, the symptom might flip between hang and wrong checksum, but it still gets triggered by the same sequence).
This seems very weird to me, and I never noticed it before (but before I only ever used the Dynamic C 8 and 9 pilots as secondary loaders), only now did I try other data as secondary loader.

spth, 2025-06-26:
Since https://github.com/spth/OpenRabbit/commit/3d55f383ef781185c5ec829b159214e770f0a917 the very basic Rabbit 5000 support is there: we can read read out some info (details see above). While we can't write the flash yet, the currently working parts should be enough to read out the flash on Rabbit 5000 - based systems once we have readout support in OpenRabbit (i.e. for readout, there are no Rabbit 5000 - specific missing bits).