spth/OpenRabbit #29: Leverage OpenRabbitRFU codebase to download flash contents

tomlogic, 2025-04-16:
A back burner project of mine has been to implement to bootstrapping process in Python so I could create a tool to extract the current flash contents via the Pilot BIOS's interface.

I've written a program (extract_flash.c, in the Samples of both Dynamic C Github projects) that I can compile to RAM from the command-line Dynamic C compiler that dumps the flash contents as base64-encoded data, but it's extremely slow when running as a standard program and printing data through the debug interface.

The Pilot BIOS communication protocol allows for reading any flash address, in addition to the code used to write to any flash address to allow for firmware uploads.  It should be possible to save flash contents to a local .bin file.  

I have some Python-based tools that look for a `prog_param` structure in the BIOS to identify the "high program address" (HPA), which allows for truncating a full flash dump to obtain just the installed firmware.  Another program can dump the contents of a System ID Block.  I need to clean those up and create an Open Source repository for them, but I could re-implement some of the techniques within OpenRabbit code to pull information on the attached target.

spth, 2025-04-22:
Yes, I agree that being able to read out the flash is an obvious useful feature for OpenRabbit. IMO, we should even support saving to both .bin and .ihx (like we do when writing the flash).

spth, 2025-06-19:
It looks like issuing TC_TYPE_SYSTEM / TC_SYSTEM_READ commands from OpenRabbit works, and results in the pilot returning the memory content at the physical address (only did a quick experiment reading RAM, and checked that I was able to make the pilot read itself, and report it back to OpenRabbit).